bofh69/spoolman2slicer #3: Import boolean variables

alex0815-bit, 2024-10-14:
Hi. Don't know if this is an issue or not:

In templates I can not declare boolean variables:

> "enable_pressure_advance": [
>         "{{extra.pa_active|default(0)|bool}}"
> ],

or 

> "enable_pressure_advance": [
>         "{{extra.pa_active|default(0)|boolean}}"
> ],

do not work. The workaround is to create the extra fields in spoolman as integer and declare them in sm2ss as int.

I'm using spoolman 0.20.0 on Debian, OrcaSlicer and sm2ss on windows. 
Thanks for your tool -  that way I only have to create filaments once and all of the data is imported into the slicer ;)

bofh69, 2024-10-17:
Hi!

jinja2 is used for the template rendering. It has no built in filter to convert an integer to a boolean, but you can use its if-statements for it. You can try something like this:
```
{% if extra.pa_active != 0 %}true{% else %}false{% endif %}
```

I've tried this with the jinja2 and there it worked, but I've not yet tested it with sm2s.

Jinja's documentation is here:
https://jinja.palletsprojects.com/en/3.1.x/templates/

It is not so well written IMO, but everything is documented there.

bofh69, 2024-10-17:
Now I've tried it with spoolman2slicer too. It seems it wants "1" as true, "0" as false.

So I'm using this in my template now:
```
{% if extra.pressure_advance %}
    "enable_pressure_advance": [
	"1"
    ],
    "pressure_advance": [
        "{{extra.pressure_advance|default(0)|float}}"
    ],
{% endif %}
```

alex0815-bit, 2024-10-17:
That's it - Thank you.

closed 2024-10-17 by alex0815-bit