bofh69/spoolman2slicer #7: Spool ID is not correct because there is a filament ID

MrTomek, 2024-10-23:
Great job @marek Simple and useful project until I implement spoolman for Orca.(https://github.com/SoftFever/OrcaSlicer/issues/2955). 

Back on topic, the templates load the ID of the filament, not the spool, which can cause a problem if someone has several spools for one filament, then the IDs are not the same and it becomes a problem. 

PS. I have a feeling that it can be done a little differently, looking at this importer. By loading all the spools, and then filtering them so that only one spool (the one with the least amount of filament) of each type of filament is loaded. I'm just throwing this idea out there like this, because it's just a preliminary idea and has a few problems to solve in order to implement it well

bofh69, 2024-10-24:
It is intentionally that it uses the filament's ID. The filament's gcode in the slicer sets the wanted filament type and when the print starts to use that filament, it will pause the print and wait for a spool change if the wrong filament is loaded, but any spool will do.

I update the actual loaded spool in klipper/moonraker by another program, [github.com/bofh69/nfc2klipper](nfc2klipper), but it can also be set in mainsail etc when the spool is changed.

I guess I could add a mode to generate filaments per spool instead, but then you would get into trouble if you do multicolor prints and use up the last filament on a spool.

scottessner, 2025-01-30:
@MrTomek I think the answer you seek is in bofh69's other tool **nfc2klipper** that he mentions.  He has a macro in that repo called `ASSERT_ACTIVE_FILAMENT` that checks that the filament ID in this template matches the filament ID for the active spool in klipper.  

@bofh69 That macro is quite clever and very valuable.  It would be helpful if it were more prominently highlighted in this repo as this tool depends on it to function as designed.

bofh69, 2025-01-30:
@scottessner good idea! I'll add it to the readme.

bofh69, 2025-02-02:
I've make a program, spool2klipper, that can be used to set the active filament when the spool id is changed in moonraker. Even if using nfc2klipper, its useful if the tag doesn't work for some reason and you change the spool in the frontend etc.

Beagon, 2025-07-27:
> For me to clarify: I would set the filament type with spoolman2slicer with "Filament Start G-Code". If it is incorrect, I would select the correct filament in i.e. Fluidd and then if there is a spool change, spool2klipper would take care about this.
> 
> Is this correct?

I would like to know the same, I'm trying to integrate this into my MMU solution as well. The last piece of the puzzle is to get spoolman to know what filament is loaded automatically. The MMU has an RFID reader and knows the filament, spoolman just needs to recognize it...

Beagon, 2025-08-04:
I have fixed this for now by forking Moonraker and adding my own function. You can give a filament id and it will get the most used spool from it. Not a perfect solution but better than nothing: https://github.com/Beagon/moonraker/tree/spoolman-by-filament-id