bofh69/spoolman2slicer #15: Multiple output variants in a single spoolman2slicer run

fweber3, 2025-03-26:
Sorry for the length, I want my usecase to be clear.

### Problem description

Some of my filaments need different settings not just per filament, but also per printer. One example among several:
- On printerX, TPU1 needs a retraction of 5mm while TPU2 needs 5.5mm [not actual numbers]
- On printerY, TPU1 needs a retraction of 6mm while TPU2 needs 6.2mm [not actual numbers]
- Actual values are purely empirical and can not be derived from other information, notably:
"printerY always 1mm more than printerY" is incorrect and
"TPU2 always .5mm more than TPU2" is also incorrect.

I made a pure slicer-config setup work in the past, but it is a major headache to keep it internally consistent. spoolman2slicer already helps _a lot_ with that, thank you! 🥳 

### Current solution

Currently I'm storing all data in Spoolman, using extra fields. I generate slicer configs using different spoolman2slicer presets like "prusaslicer-all-anycubic", "prusaslicer-modified-dev-tronxy". These presets have
- a printer specific filename template like `MATERIAL_COLOR@Tronxy`
- file templates that are mostly identical (yay, maintenance!) but differ in a few crucial blocks like `retraction = {extra.get("filament_retract_lift_tronxy")}; compatible_printer_condition = Tronxy ...`

I then run spoolman2slicer with the same output directory but differing templates. The Prusaslicer UI later filters all filaments that do not fit the currently selected printer.

From a slicer user's perspective this works great. From a filament admin's perspective it is so-so. However, abusing the preset system like this is Not Very Pretty (tm) and requires constant local hacking when updating spoolman2slicer.

### Feature request

I would like the ability to
- generate multiple (sets of) output files from a single (set of) templates
- with one (or more) variables available inside the template that tell me which set is being run
- where the variable(s) can be specified at the commandline, or even in a config file

Examples:
- `./spoolman2slicer --slicer=prusaslicer --variant=A ...`
--> runs the `prusaslicer` template once, with the variable `variant` set to `A`
with template file containing `... {% if variant == "A" %} ...`
- `./spoolman2slicer --slicer=prusaslicer ...`
with user-specific config file containing `variant = A|B|C|D`
and template file containing `... {% if variant == "A" %} ...`
--> runs through the `prusaslicer` template multiple times, with the variable `variant` set to `A`, `B`, `C`, `D` successively
- ideally, this would work together with `--updates` mode, but that's just be the icing on my cake 🥇

Or any other mechanism with the same end use?
Or am I overlooking something obvious??

Thanks for your consideration!

bofh69, 2025-03-29:
Thanks!, its a great idea!

I've created a branch and pull request for it now.

Let me know if that works for you.

fweber3, 2025-03-29:
Works beautifully, first time, including --updates!

Remarks:
1. I suggest renaming the CLI argument from "variant" to "variants". If used at all, it'll probably be more than one.
2. Could I get the URL as additional variable?

bofh69, 2025-03-30:
Right, it should have been variants. I've fixed it now and added a spoolman_url field to sm2s.

fweber3, 2025-03-30:
Thanks a lot!
Already implemented in my workflow 😀

PR #16 (2025-03-30): Add variant argument
It can be used to generate one template per printer in multi-printer setups etc.

Fixes #15
commit 4283b141 (2025-03-30): Add variant argument

It can be used to generate one template per printer in multi-printer
setups etc.

Fixes #15