bofh69/spoolman2slicer #17: Windows version of Orca Slicer 2.3.0

juhjar, 2025-04-15:
Hello,

I managed to make output to NFS shared folder on NAS. .json and ,info files are generated, but how can I import them to Orca Slicer? 

-juha

bofh69, 2025-04-16:
The files are not possible to import, the program is designed to directly write the slicer's configuration files to its directory for filament configuration.

I don't have any computer with Windows so I can't test it myself, but I would guess you have the config files for filaments somewhere under C:\Users\%USERNAME%\AppData\Roaming\OrcaSlicer\ or maybe Local instead of Roaming. Then it is probably in user\default\filament\.

You should probably not use my template files directly, they are most likely not right for your setup, but instead create your own templates from the existing filament configuration files and modify them like how my template files have been modified.  My files might work, but I'd guess you would at least have to modify the "inherits" setting and the "filament_start_gcode".

bofh69, 2025-04-16:
Btw, you should be able to run the program directly on your windows computer. You just need to install python3 first.

closed 2025-10-02 by bofh69