bofh69/spoolman2slicer #21: Hardcoded username in config_dir variable

darkreeper, 2025-05-01:
Hi @bofh69,

I just saw your username hardcoded in the config_dir variable.
When I was running the py file I was wondering why it was always using your username.

`config_dir = user_config_dir("spoolman2slicer", "bofh69")`

This should be taken from the os as logged in user or something.

Best regards,
Christian

darkreeper, 2025-05-01:
I have to rename this. 

For Windows it should look like this:

`config_dir = user_config_dir(appname="spoolman2slicer", appauthor=False, roaming=True)`

bofh69, 2025-05-01:
Thanks!

I thought it was the vendor name. I should have read the documentation before changing it... 

I'll fix it.

darkreeper, 2025-05-02:
According to documentation it is the vendor name.
Unfortunately, this results in a very strange directory chain construc, where the "appauthor" is used again, something like so:
c:\user\crosc\appdata\local\spoolman2slicer\bofh69\user\default\filament

PR #22 (2025-05-02): Fix config dir on windows
Fixes #21
closed 2025-05-02 by bofh69